WorldWideTelescope/wwt-windows-client #192: Jagged lines in 6.0.901.0

astrodavid10, 2021-12-30:
Version 6.0.901.0 shows jagged lines upon opening the application that resolve themselves with a window resize.
[image]
.

pkgw, 2022-01-03:
I see that you found the work that I've been doing towards automating release builds! @Carifio24 will take a look at this. Is this issue happening on a HiDPI display?

astrodavid10, 2022-01-03:
Yes! And also yes, 4k monitor, although with Windows scaling it is 2560 by 1440

Carifio24, 2022-01-04:
I would guess that this is a resolution/display-based issue, since I'm not seeing this on the monitors on my machine (neither of which are HiDPI). I'll see if there are any settings I can tweak to try to reproduce this.

astrojonathan, 2022-01-04:
HiDPI and the modes are a difficult challenge. I can take a look at this and see if I can create a workaround to cause a immediate rescale on startup to let DX know the DPI scaling is in place. I think it has to do with initialization flow in the HiDPI scaling on startup.

Carifio24, 2022-01-14:
I've been trying to look into this, but without access to a HiDPI monitor it's pretty difficult to test anything out.

If a window resize makes the jagged lines go away, would calling `OnResize(EventArgs.Empty)` in the `Earth3d` constructor (or maybe `Earth3d_Load`) do the trick? Obviously it would be nice to find a real fix, but if we're somewhat pressed for time, maybe that's a workaround.

astrodavid10, 2022-01-16:
This was an issue in the past #131 and #132

Carifio24, 2022-01-27:
Here's a build with a call to `OnResize(EventArgs.Empty)` at the end of the `Earth3d` constructor. @astrodavid10, could you (or anyone else who has a HiDPI monitor) try this out and see whether it fixes the problem?

https://drive.google.com/file/d/11I3Wx5bbz0E1Aoh8IunBt5ykGHFiGREM/view?usp=sharing

astrodavid10, 2022-01-28:
@Carifio24 This checks out for me - no more jagged lines!

Carifio24, 2022-01-28:
Great! Then we can use this workaround until we come up with something that more directly addresses the root cause.