WorldWideTelescope/wwt-windows-client #196: Crash with Intel Iris Xe graphics

pkgw, 2022-01-06:
A user reports a "WWT has encountered an error" crash on the following hardware:

- Intel Iris Xe graphics (4 GB)
- DirectX 12
- New Dell Inspiron i7

Crash is:

> HRESULT: [0x887A0005], Module: [SharpDX.DXGI], ApiCode: [DXGI_ERROR_DEVICE_REMOVED/DeviceRemoved], Message: The GPU device instance has been suspended. Use GetDeviceRemovedReason to determine the appropriate action.

pkgw, 2022-01-06:
Based on my initial investigations, this appears to be a driver crash. We could potentially figure out what WWT is doing that is making the driver unhappy and figure out a workaround, but that would likely take a huge amount of effort and resources that we just don't have. Hopefully we can identify a version of the driver that doesn't have this problem (which apparently triggers pretty much instantly on startup).

See [this 2021 forum thread][1] for potentially related discussion. The Intel support folks pointed out that the original reporters were using OEM-customized drivers, and recommended switching to Intel's generic drivers.

[1]: https://community.intel.com/t5/Graphics/DXGI-ERROR-DEVICE-REMOVED-Intel-Iris-Xe-Graphics/m-p/1274437

[This specific post][2] gives the recipe for installing Intel's generic drivers. However, the latest available drivers are now newer, as of today the latest generic driver version is 30.0.101.1191, [available here][3]. The first thing to try would be update the drivers if they're not already at the latest version.

[2]: https://community.intel.com/t5/Graphics/DXGI-ERROR-DEVICE-REMOVED-Intel-Iris-Xe-Graphics/m-p/1274437
[3]: https://www.intel.com/content/www/us/en/download/19344/intel-graphics-windows-dch-drivers.html

pkgw, 2022-01-07:
Unfortunately, Intel generic drivers with versions ranging between 26.20.100.7985 and the latest, 30.0.101.1191, don't help. We'll see if Intel's support can offer any help, although frankly I'm not too optimistic.

pkgw, 2022-01-12:
If anyone here has the relevant hardware, it looks like the most promising place to pursue this issue with Intel might be their [Graphics community support forum](https://community.intel.com/t5/Graphics/bd-p/graphics).

pkgw, 2022-01-18:
Also reported in a Dell Inspiron 7706 2n1 ... maybe. The machine in question also has a discrete graphics card, so you'd expect/hope that WWT would be using it rather than the Intel, but the nature of the crash is identical to the one reported above.

pkgw, 2022-01-20:
Report with the 2-in-1 followed up to confirm that switching the preferred GPU to force the use of the nVidia made the problem go away.

IceyOrange, 2022-10-12:
I met the same problem and unfortunately choose another GPU is the only solution of this problem...