WorldWideTelescope/wwt-windows-client #198: Pasted spherical layer data adds a depth parameter incorrectly that cannot be removed

astrodavid10, 2022-01-15:
I attempted to paste formatted CSV data to create a layer of crater positional data in Lon/Lat (no elevation parameter) on Ceres, which was added as a new reference frame orbiting the Sun (as a child of the Sun). The resulting layer seems to make use of a distance parameter which I do not see active anywhere in properties. I have tried to change all the properties individually and do not see any way to remove the depth parameter. If I paste the same data onto a "first class" reference frame like the Moon, this does not occur and everything works. I think this is the result of something tied to "second class" reference frames.

The WWTL file is attached and a WTML for Ceres can be found here: https://data1.wwtassets.org/packages/2022/01_ceres/Ceres_Dawn_ClrSh_Oct_2016/ceres_clrsh_v1.wtml 
[ceres_craters_v1.zip](https://github.com/WorldWideTelescope/wwt-windows-client/files/7874240/ceres_craters_v1.zip)

Expected outcome (achieved with a KML):
[image]

Correct Outcome on Moon:
[image]

zoomed out broken view:
[image]

[image]

Zoomed in broken view:
[image]

pkgw, 2022-01-17:
The depth parameter must be getting associated with one of the spatial coordinates — that's how you'd get the coherent looping structures. Maybe something exposed by the work in #181?

Carifio24, 2022-01-27:
I've been looking into this recently, and I think I'm making some progress. I can say that it's not from enabling the distance functionality for VO table layers, since the layer in question here is a SpreadsheetLayer.

@astrodavid10 What should the mean radius of this reference frame be? When I load it in, the mean radius is 10 meters? Should it be the radius of Ceres (469730 m)?

Something I've noticed, that I think may be related to this problem, is that changing the radius of a reference frame does not cause that frame's layers to be redrawn (the layer isn't marked as "dirty"). It seems that the point distances get scaled by the appropriate amount, but for non-astronomical layers like this one, the "altitude" coordinate depends on the radius (e.g. [here](https://github.com/WorldWideTelescope/wwt-windows-client/blob/master/WWTExplorer3d/SpreadSheetLayer.cs#L1295)), so that's not enough.

As an example, after I open the Ceres layer that David sent, I get:
[image]

If I change the reference frame radius to 469730 (the radius of Ceres), and zoom out, I get:
[image]

So the same shape, just larger. But if I do something that forces the layer to be marked as "dirty", like opening and closing the layer properties (without changing anything), the display changes to:
[image]

Just for reference, after zooming in a bit and changing the markers to use the size attribute:
[image]

astrodavid10, 2022-01-28:
Hey that looks much more on target! Yes, the mean radius should be that of Ceres, so 469730 m.