WorldWideTelescope/wwt-windows-client #200: HiPS used in a tour is not recognized after tour is reopened

astrodavid10, 2022-01-16:
Upon setting a HiPS dataset as background or foreground imagery for use in a tour, the Winodws client will accept this and playback as normal. If saved to run on the web client, all works as intended see here: https://bit.ly/wwt-orion_dss_unwise but, if reopened in the Windows client, the HiPS dataset appears black.

pkgw, 2022-01-17:
Just to double-check what you mean ... you're creating a tour in Windows, opening it on the web, re-saving it out of the web client, and then re-opening on Windows?

astrodavid10, 2022-01-18:
Creating tour on web, opening in Windows (doesn't work), edit then save in Windows (works), reopen second version in Windows again (does not work.)