WorldWideTelescope/wwt-windows-client #201: "Wait for all downloads" in Render to Video does not wait for all downloads

astrodavid10, 2022-01-16:
I am not certain if selecting "Wait for all downloads" when rendering a tour to video (really frames) actually works in any instance, but it definitely does not if a HiPS dataset is the one being rendered.

astrodavid10, 2022-01-16:
I may have jumped the gun on what to call this... in looking at my render, the data tiles in at the appropriate zoom level and then momentarily backtracks and drops a chunk of the view to a much lower resolution. This is in a 3840x2160 screen size at 100% Windows scaling and at 60 fps

pkgw, 2022-01-17:
I think I get what you're describing. Does the resolution glitch last, like a single frame, or longer than that?

My guess is that this one might be tricky to trigger so if you're able to report any more notes about how to reproduce the issue consistently, that would probably be very helpful.

astrodavid10, 2022-01-18:
I don't think that this is only a HiPS problem, but it might be HiPS exacerbated. In this render, it lasts 40 frames which is 2/3 of a sec.
Pre glitch:
[image]
Glitch:
[image]
.

astrodavid10, 2022-01-18:
I'll try a couple re-renders to see if it is reproducible and always in the same spot.

pkgw, 2022-01-18:
Interesting! CC @imbasimba if you happen to have the chance to think about why this might be happening.