WorldWideTelescope/wwt-windows-client #203: Add layer property to snap data to surface

astrodavid10, 2022-01-18:
Currently pasted layer data will billboard towards the camera - it would be great to add an option to snap to the surface of the reference frame that it is attached to.

pkgw, 2022-01-18:
Based on one of your other issues, KML imports don't have this problem, right? If that's the case it would at least be reassuring that maybe it wouldn't require major surgery to fix this issue.

astrodavid10, 2022-01-28:
KML imports do snap to the surface!

pkgw, 2022-01-28:
That's good to know. Would life be easier if we set up some tools to help create and edit KML data? Maybe that's the easiest path forward for this kind of stuff. Although I think that the webclient doesn't include any of the KML code ...

astrodavid10, 2022-02-01:
Yes definitely, which leads to bummer...

pkgw, 2022-02-01:
It would be a good research project to look into the Windows KML implementation and think about what it would take to port it to the web. Depending on the amount of sophistication in the existing codebase, it might make more sense to port it over, or maybe it would be easier to add some more bells and whistles to the existing spreadsheet code.