WorldWideTelescope/wwt-windows-client #209: crash on startup

xalspaero, 2022-04-11:
device: surface pro 8
OS: win 11
app version:  6.1.1.0

I run the app, and immediately get this error box:

AAS worldwide telescope has encountered an error.  click ok to restart or cancel to attempt to ignore the error and continue.

pkgw, 2022-04-22:
Hello,

My apologies for the slow follow-up here. What kind of graphics card do you have? The only known issue we have that leads to this is a driver incompatibility with Intel Iris Xe graphics, as discussed here: https://github.com/WorldWideTelescope/wwt-windows-client/issues/196.

xalspaero, 2022-04-22:
I have the one you mentioned.  Stock SP8 whatever.  Device manager: Iris Xe

pkgw, 2022-04-22:
Unfortunately we don't have a solution for this issue yet. If your computer has an additional discrete GPU, it should be possible to get things to work by forcing the OS to use that card for WWT rather than the Iris Xe graphics. If not, we're not aware of any effective workarounds, and I'm afraid we don't have the resources to figure out the reason for the driver crash and work around it. My best recommendation is to file a report with Intel in their [Graphics community support forum](https://community.intel.com/t5/Graphics/bd-p/graphics), and hopefully they will eventually issue a new driver that fixes the issue.

I am going to close this issue as a duplicate of #196 — we can continue the discussion there as needed. That way the investigation of this problem doesn't get splintered across a bunch of issue reports.

closed 2022-04-22 by pkgw

xalspaero, 2022-04-22:
Understood.  Glad it is being tracked elsewhere.  Hope my surface can use this app someday...