WorldWideTelescope/wwt-windows-client #210: GotoSaturn binding not working

SCfmk, 2022-04-29:
When I use the gotoSaturn binding, either with a view-button or through a midi controller, I get a switch to skyMode and an overlay with Saturn's dust clouds from Spitzer in stead. I'm using 6.1.1.0 and have tried this on both a windows 10 and a windows 11 (both 64bit)
[image]

pkgw, 2022-04-29:
Thank you for the very clear issue report! It looks like the code that looks up "Saturn" gets confused because this image in our database has exactly that name as well. I think we should be able to fix this with a data update that renames the image. It might make sense to remove the image altogether since it looks like it might not really map onto the sky in a sensible way, anyway.

AstroRob, 2022-04-29:
Hi,

Peter, as the creator of this image I agree that this image should probably just be removed… I don’t think it even had WCS metadata embedded so am surprised it showed up at all.

Since this was a solar system observation, its coordinate only makes sense with respect to Saturn’s location at the time of observation, and only from Spitzer’s viewing geometry in space at that time… so not really meaningful as a sky overlay.

Cheers,
Robert

Robert L. Hurt, Ph.D (he/him) | Visualization Scientist
Caltech/IPAC | IPAC Communications and Education
Mail Code 314-6 | Pasadena CA 91125